niclet/yi-hack-v2 #1: Yi hack v2 collaboration

ant-1, 2016-07-13:
Hi Niclet!

Thank you for sharing your work !
I'd like to help you working on it but I can't find where to buy the Xiaomi Yi v2 (in France or Europe)
Do you have any tips to buy it ?

Sorry to contact you this way, I know it's not the right place but I didn't find your email anywhere

TommyChausson, 2016-07-13:
Hi,

Just in case, Personally  i bougth mine from Aliexpress http://www.aliexpress.com/item/100-Original-Xiaomi-Yi-Smart-Camera-2-1080P-Full-HD-Xiaoyi-ants-Smart-Webcam-2-IP/32654699860.html?spm=2114.40010208.4.9.7m2jin

I'm not sure that you can find it in europe

Regards,

niclet, 2016-07-13:
I bought mine on http://eud.dx.com
But even if it is the EU site, it comes from China and it is the chinese version. I don't think that there is an international nor european version of the V2 yet, it is available only on the chinese version of the Yi Home iOS Application.

Regards.

closed 2016-07-13 by niclet