niclet/yi-hack-v2 #2: Any chance to get a snapshot from the camera?

hostmit, 2016-08-16:
Hello! Any chance to implement snapshot from camera (photo). I need a way to get a photo when needed. Over telnet command or cgi script?

Please help.

niclet, 2016-08-16:
Hi,
I'll have a look at this request asap.
I may have already found such kind of tool on the camera. Don't remember exactly.

hostmit, 2016-08-16:
Thanks for quick response.
Already went through filesystem of the camera - found only mp4record and rtspsvr, which apeears to fetch data directly from sensor. I wish we had a sources for these. :(

niclet, 2016-08-19:
I got it working. Will provide a small sh asap.

niclet, 2016-08-19:
Could you try test/v2/scripts/capture.sh ?

hostmit, 2016-08-19:
Thanks for getting back to me.

I came here from  fritz-smh/yi-hack, was going to try your flash but have some doubts.
Yours "home.bin", not "home" without extension. Should we rename the firmware (otherwise it wont flash itself)? Also size is a lot different, ~8MB vs 32MB. I do have non chinese version. Is it safe to flash? :)

niclet, 2016-08-19:
If your camera is not chinese, it is probably the first version, not the v2 which I manage here. The v2 has its microphone on the white support around the black camera.
The firmware of the v2 is really home.bin and its format is different from v1 which explains different sizes.
If you already used fritz-smh hack, then your camera is v1 and don't use this one.

hostmit, 2016-08-19:
Thanks. Can I have the binaries of test_yuvcap and jpg_enc you are using in your script? I'd like to test it with V1.

niclet, 2016-08-19:
[Archive.zip](https://github.com/niclet/yi-hack-v2/files/427418/Archive.zip)
I don't think that processor architecture is compatible between v1 and v2, but you can test :)

hostmit, 2016-08-19:
# cat /proc/cpuinfo
Processor       : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 218.72
Features        : swp half thumb fastmult edsp java
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : hi3518
Revision        : 0000
Serial          : 0000000000000000

What do you have @ V2? Also where from are these binaries? Do you have a source?

niclet, 2016-08-20:
**cat /proc/cpuinfo**
---
processor       : 0
model name      : ARMv7 Processor rev 1 (v7l)
BogoMIPS        : 597.60
Features        : swp half thumb fastmult vfp edsp thumbee neon vfpv3 tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x4
CPU part        : 0xc09
CPU revision    : 1

Hardware        : Ambarella S2L (Flattened Device Tree)
Revision        : 0000
Serial          : 0000000000000000

---

The binaries I use in my script are from the camera itself, they are part of the firmware. And obviously, there is no source for them.
When I need to compile my own binary, I use a Raspberry Pi 3, which uses /lib/ld-linux-armhf.so.3 too.

closed 2016-08-26 by niclet