niclet/yi-hack-v2 #3: How to change wpa supplicant?

cuibap1212, 2016-10-10:
Hello, 

I want to change wpa tkip security to wpa-aes or wap2-tkip, wap2-aes.
How can you change it?

niclet, 2016-10-10:
Hello,

I have no such Wifi to check it for now, but the wpa_supplicant.conf is a standard on linux, so it might be usable.
Otherwise, if you succeed in connecting to your wifi with the official Yi Home application, you should then be able to find the official wpa_supplicant.conf on the camera through telnet access. I don't remember exactly where it is located but something like this should tell you :

find / -name wpa_supplicant.conf

Regards.

cuibap1212, 2016-10-10:
Hi @niclet ,

Could you please step by step to find wpa_supplicant.conf on sd?

Thanks,

sebastianrothe, 2017-09-28:
The file is on your sdcard under test/wpa_supplicant.conf

The WPA2 configuration has to be like this:
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="yourssid"
scan_ssid=1
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP
group=CCMP
psk="yourpassword"
}