niclet/yi-hack-v2 #6: Night vision

Prockkaii, 2016-11-22:
Hello, 

I have the international version but the night vision isn't working. 
Do you have this issue? 

I try to use both latest version (us and eu), but it's still the same. 

Thank you

Prockkaii, 2016-11-22:
I notice another issue with colors.
For example, green color turn out red.

Do you have the meaning of parameters you had use on command **/usr/local/bin/test_encode** ?

Do you know which command make the RTSP server and Yi home app incompatible together ?

In fact, I try to use this camera with Synology.
On one hand, I like the Yi home app : connection is really fast and I like the "phone chat" ability (you can chat with the person in front of the camera).
On the other hand, I like Synology for it's cloud function. But it's still not working very well : video seems to be not in full HD even if there are in 1080p and files are too big.

niclet, 2016-11-23:
Sorry, I don't have time anymore to work on it for now. Hope to be back later.
From telnet access, you can call test_encode without arguments, it should print help and samples.

closed 2016-11-23 by niclet

TommyChausson, 2016-11-28:
Hi @Prockkaii , did you success to solve the night vision issue ?

Prockkaii, 2016-11-28:
Hi, 

Nop, in fact using the rtsp protocol with surveillance station is not a good solution. 
Files are very big.. So the time, files are in the cloud. You can't catch thief.. And also you can't use the mi's app : you can talk with users! 

For now I'm using the camera with the ftp server enabled, and the synology is scanning every minutes of their is or not new records. And if there is, there are uploaded into the cloud.

TommyChausson, 2017-01-12:
A little thing about that ... if somebody want to try ir working there is a script wich can set or unset the ir cut

`/bak/usr/script/ircut close`

`/bak/usr/script/ircut open`

fcserg, 2017-02-13:
Hi, TommyChausson
> if somebody want to try ir working there is a script wich can set or unset the ir cut
I tried to launch this script, rtsp image changes, but this is not that night vision, we got in application.

TommyChausson, 2017-02-14:
@fcserg  Yes i saw that :( i did'nt had the time to test it in night condition when i wrote it, it seem's that i's a wrong way ...

fcserg, 2017-02-14:
My colleagues suggests that it could be night mode, but not IR backlight, so we need to find how to enable IR backlight.