niclet/yi-hack-v2 #7: Flux rtsp/Audio

TommyChausson, 2016-11-26:
Bonjour Niclet,

Le flux RTSP est fonctionnel j'ai réussi à le brancher sur mon NAS Synology, je me suis aperçu que l'audio était manquant, en l'ouvrant dans VLC je constate également que ce dernier est peu bruyant :) , est-ce que le flux est censé intégrer l'audio d'entrée de jeu ?

TommyChausson, 2016-11-28:
Je n'avais pas vu les précédentes réponses sur ton manque de disponibilité, désolé,

TommyChausson, 2017-01-12:
The parameter "--au-type" of the test_encode looks like to be the good one 
`--au-type [0~3]              0: No AUD, No SEI; 1: AUD before SPS, PPS, with SEI; 2: AUD after SPS, PPS, with SEI; 3: No AUD, with SEI.`

My test on it doesn't seems to work, anybody tried ?

niclet, 2017-01-12:
Nice try, but this parameter seems to be specific to H264.
That's what I understand when looking for AUD, SEI, SPS, PPS on google.
As far as I know, audio and video are managed by two different processes when using camera "normally". So we would have to post process rtsp stream to include audio.
I had success to record audio on the camera, but multiplexing it with video is another story.

Starfine, 2017-01-19:
I have same problem. I Used Synology Surveillance Station APP, it can watch video but no sound too.
Thx!