niclet/yi-hack-v2 #9: Yi2 1080p camera doesn't work anymore outside of China :(

Comraddm, 2016-12-17:
Hello!
10 days already since the camera stopped to show live video :(
Not only my camera, many people say the same.
Do you have the same problem?

amiraetari, 2016-12-18:
Hi,
I have the same problem.
Yi Home app don't pair with camera and show following error:
"Network connection failed, please check network setting and reconnect it (-20009)"
Any solution?

niclet, 2016-12-18:
Hi,

I can't check on my side for now, but many people have same issue. 
A friend of mine has the v1 (780p) and yihome app doesn't work anymore, he must use mihome app. But as far as I know, mihome doesn't support v2 on iOS.

amiraetari, 2016-12-18:
I've also checked with Mi Home app but don't work same as Yi Home.

niclet, 2016-12-18:
For v1, even MiHome app doesn't work often correctly. MIHome crashes for universal remote. XIaomi connected LED has burned after one day... Seems xiaomi quality is not on the right way.

niclet, 2016-12-19:
Actually, Xiaomi may have upgraded their servers to force us to upgrade camera firmware. Are you using original old firmware ?

Comraddm, 2016-12-19:
Yes, we all use old firmware.
Don't you have Yi2 now, why can't you check it?

amiraetari, 2016-12-19:
I'm also using old firmware (2.1.1_20160429113900) and had no problem until 10 days before.
[image]

niclet, 2016-12-19:
Since I don't have enough time to work on these projects for now, I have temporarily given my cameras to a colleague, that's why I can't test it myself.

Comraddm, 2016-12-19:
Can you at least tell where to download the newest firmware for it?

Comraddm, 2016-12-20:
Tested the latest CN, USA and EUR firmware with Chinese app and with English app, but without success.
Every time it says the camera can't work outside of China.
Only the old April's firmware works, but still no live video from the camera.

Comraddm, 2016-12-20:
Also tested your hack with the camera, it works like it should.
Opens FTP, Telnet and RTSP flow.
As I understand, when I switch on RTSP flow, the camera stops writing files to the SD card?
Can it work simultaneously RTSP flow and SD card files writing?
I want to watch live video by another app and copy video files from camera by FTP.

TommyChausson, 2016-12-20:
It's the /home/web/ipc script who makes all automatics actions : 
-> Detect Move and save to sd card
-> Adjust image (night vision included)
-> send informations to external servers

When you launch the startup_modified, this script is not launched... that's why you cannot (for the moment i hope...) do the both at the same time...

niclet, 2016-12-20:
Both can't run at the same time. The camera can probably be accessed by only one process at a time. 
Sorry, I don't have time anymore to keep on working on these projects, but you're welcome to improve them :)

ykhandler, 2016-12-21:
Hi niclet. If you dont mind, Can we get source code of the libyihackv2.so? So we can  improve them...

atmirr, 2016-12-24:
Any recent solution guys?

atmirr, 2017-01-09:
Dear @niclet 
We're still have the issue, it's good to know that our 50 pcs of these cameras stop working.
We're even ready to pay you to fix this problem.
Please keep in touch with me.
Thank you

niclet, 2017-01-09:
Hey,
Thanks for proposal, but I don't do that for money :)
My children are taking all my time.
It would be interesting to check if an "international" version of the camera still works outside China.
And if someone owns one, we could dump all the camera files and find what is the difference between a Chinese and an international version.
I already know that the serial number contains a country identifier, would need a valid international serial to check. 
As far as I know the firmware is the same for both Chinese and international versions.

ykhandler, 2017-01-09:
I can say that the international version works well. How can i tell that? I got xiaoyi dome china and international. The day ants2 got -20009 error, dome also got the same error. While the dome international still working. As i know, there is nothing difference between the hardware and software in the same kernel.

niclet, 2017-01-09:
So it's probably only the serial number which is different. We can change the serial number of the camera, but at the time I did it, it was not working at all, as if Xiaomi was able to check if serials are valid.

ykhandler, 2017-01-09:
You got that right. Not only serial is the validation. There are 2 other key is for the validation. I hope i could know how it was generated. 
The best solution is to create our own program. So we can use it over ddns than depends on other server.

ashmodai6, 2017-01-19:
I do own both a Chinese and an International version of this camera. Happy to provide any information needed to help making this cam Yi-independent ;) Just tell me what you need and how to get it

atmirr, 2017-01-20:
I also have 2 Chinese and International too.
Ready for any experiment.

jasoon88, 2017-01-20:
I just ordered 4 Xiaomi Yi 1080p from china, i loaded the hack firmware to remove "camera can only be used in china message" Successfully added camera to the chinese Yi app but i can never get the camera to load image. "Network connection failed, please check network setting and reconnect it (-20009)"
However when i check the sd card i can see video clips of the camera stored i guess via motion detection.
Can anyone tell me how to get it running with the app??!

AlexKholodkov, 2017-01-21:
unfortunately moved to China, will be there to work with the original application only

xmflsct, 2017-02-16:
@atmirr any luck till now?

atmirr, 2017-02-28:
@xmflsct unfortunately not. It seem that someone working on it here:
https://github.com/fritz-smh/yi-hack/issues/141

xmflsct, 2017-02-28:
Thanks @atmirr! I will follow that thread.

ykhandler, 2017-07-24:
Check this out:
https://m.youtube.com/watch?v=HssoXD_U7eA

jasoon88, 2017-07-24:
How did you do it?

Thanks,
Jason Liu

Reply-to: reply@reply.github.com
Cc: jliu.88@hotmail.com; mention@noreply.github.com

https://m.youtube.com/watch?v=HssoXD_U7eA

ykhandler, 2017-07-24:
Just create a custom firmware. But i'm sorry i cannot give it for free, because i need money and i sell services in my country.... Really2 sorry.

fjkum, 2017-07-24:
Why everyone still using the China version of it when there is an International version of it from the official Yi.

niclet, 2017-07-24:
@ykhandler 
Ok, so quite useless here on github, and as said by @fjkum, people should buy international version now.

ykhandler, 2017-07-24:
Well, they already bought it, and they cannot return it to the seller right? 
Do you know that xiaoyi official didn't sell parts for the camera? Maybe we must not buy xiaoyi product afterall

fjkum, 2017-07-24:
Well, if that is the case, they can always do a search on YouTube, plenty of resources on how to do it for FREE.

ykhandler, 2017-07-24:
That is so right. Btw, I'm not forcing anybody to buy my service. just in case if anybody need my service, they can contact me... thanks

fjkum, 2017-07-24:
Sure. Again, this might not be the right place to do that. 
We shouldl help those ppl who needs it for NO FEEs involve.

So guys, here are the steps:
Steps to fix it.

1. Turn on the camera and hold down the reset button for 6-8 seconds

2. Turn off the camera 

3. Insert the memory card

4. Turn on the camera and wait for about 5 minutes 
(will flash orange, then turn orange, then speak Chinese and then will flash orange) 

5. Connect the camera via the camera app

Memory card
1. format the sd in FAT32
2. extract and copy the firmware file (a file called 'home') to the memory card root folder

Firmware used in this example 1.8.5.1K_201508311131 (Download at http://xiaoyi.querex.be/1.8.5.1K_2015...)

If your camera is unable to be flashed to 1.8.5.1K, you might need to flash to an older version and flash it back up to the last available working version. (so you will need to flash it twice)
Download and flash the two firmwares in the following order.
Updated: 28 April 2017
1.8.5.1B (http://adf.ly/14420685/1851b-20151321...)
1.8.6.1B (https://sites.google.com/a/mobileai.t...)

When pairing with your WiFi, please do not connect to a 5GHz WiFi Band as older versions of the camera does not support it. You will get a password error message when you try to connect to 5GHz Band.

ykhandler, 2017-07-24:
well, that is not for ants2 -1080p. Better search again bro...

niclet, 2017-07-24:
@ykhandler 
We have clearly understood that you claim to provide commercial support to hack non international cameras which are blocked outside China. This is not the aim of this repository, especially on GitHub.

fjkum, 2017-07-24:
@niclet I totally agreed!
Hope to see more features out from your source.
I am looking forward to RTSP

niclet, 2017-07-24:
@fjkum 
This is exactly the method exposed in the main page of this repository ;)

fjkum, 2017-07-24:
@niclet, ah! Silly me! :D

fjkum, 2017-07-24:
@niclet, will this still work on the latest official firmware?

niclet, 2017-07-24:
Unfortunately, I can't certify it.
Maybe new firwmares don't allow to downgrade, or maybe Xiaomi now sells cameras which are not exactly the same as mine when creating this repo.

fjkum, 2017-07-24:
@niclet, thanks for clarifying.
Can we hook up USB to Serial for accessing to debug ?

niclet, 2017-07-24:
I'm afraid USB is only available to supply power, but don't know how to be sure.

ykhandler82, 2017-07-24:
You don't need to banned me @niclet. 
I will say, even you use the latest firmware, you still able to downgrade the firmware.

niclet, 2017-08-18:
You can now make your camera work outside China if you find a working proxy in China.
You just have to value the YI_HACK_PROXY variable in the configuration file
Format is: YI_HACK_PROXY=socks5h://127.0.0.1:1080
You will have to remove the camera from your application and make association again.

niclet, 2017-08-25:
I'm currently working on a way to make the camera find itself a working chinese proxy.

zhen-pan, 2017-10-03:
Hi @niclet 
Thanks for working on this! Do you have any update on the chinese proxy self searching feature?
Thanks

n8henrie, 2018-08-28:
@niclet -- would love to get the source code of `libyihackv2.so`, especially since it looks like this repo is no longer under maintenance.