niclet/yi-hack-v2 #12: About the /home/web/ipc script

TommyChausson, 2017-01-12:
Hi,

I just open this topic to discuss about this process wich make all the cool stuff of the camera (adjusting image, set Ir, detect move etc...) And the not cool stuff (sending informations to  xiaomi servers).

I manage to run the both rtsp and ipc at the same time  but sometime it causing camera crashing. In a very dirty way  i did a script wich reboot the camera if it crash (https://github.com/TommyChausson/yi-hack-v2/commit/e1dc33418af208ac44f91f970cdbd1092d8caa5e) . Ok it's not a good way to do but it "works sometimes" !

I think there are two way to solve our problem : 
-> Understand what the /home/web/ipc script do, and do a "fork" of it to make it working cloudless and stop crashing
-> Understanding what /home/web/ipc script use (sesor, library etc..) to make image adjustement and make a script wich do the same job (without cloud).

I've got "limited" linux capacity and i'm not sure how to process... 

Is it A good Way  ?
Can anybody try to take a look at it ?

Regards,

niclet, 2017-01-12:
This is the global idea, and it is very time consuming, unfortunately.
On my own, I use some tools like strace/ltrace to understand what does a process.
When I need to compile something for the camera, I do it on a Raspberry PI 3. It avoids doing cross compilation which is quite annoying.

Best effort would be to replace xiaomi servers with our own servers...

TommyChausson, 2017-01-12:
ipc script let us some logs when running and i can see some things like that : 
`chip_image_api.c:load_lens_shading_cali_data(1001): [DONE] load lens shading calibration data,disable!`
`chip_image_api.c:load_lens_shading_cali_data(1049): [DONE] load lens shading calibration data! file =/usr/local/bin/shading3.bin shinding = 3`

By a little search on the web we can find that there is a SDK for ambarella card : 
https://github.com/mrtos/SS001_SDK2.6 where i can find for exemple the function name i mention before here : https://github.com/mrtos/SS001_SDK2.6/blob/master/ambarella/packages/img_mw/mw/arch_s2l/src/mw_image_priv.c

Does somebody think that we can use this SDK to develop our own program ?

ykhandler, 2017-04-22:
@TommyChausson, may i got a copy of the sdk? It is closed in mrtos github. i want to do some experiment with the sdk... please let me know.