niclet/yi-hack-v2 #13: Cannot downgrade/hack

xmflsct, 2017-01-12:
So today I have got a Yi Ant 2 1080p camera, with serial starting with 18CN. The firmware inside is 201612051401.

I try to follow the instructions, but no luck. It seems to stuck in bootloader - yellow light only. Then I try to downgrade to lower version, but it doesn't work. Anybody has suggestions?

p.s. I hacked a Yi Ant 720p before, so I have some ideas of what I am doing here. :)

xmflsct, 2017-01-12:
By the way, there is no log file generated onto sd card either..

xmflsct, 2017-01-13:
By the way, factory_test.sh does stop it from booting, even while it is empty. But how could I debug the process without opening up for serial communication? Thanks!

Prockkaii, 2017-01-20:
I have similar issue, after updating the camera it stayed at yellow light.. I fixed the issue for formating the sd card with another tool than the computer (ex : a phone). It's now ok, and I prefer to stay at the previous version to avoid another issue.

xmflsct, 2017-01-24:
@Prockkaii how did you downgrade? Did you format the SD card with a phone, then put the files in?

Prockkaii, 2017-01-25:
Just rename the file as "home.bin" and restart the camera. 
Finally I used another one, my card was damaged

wentianl20, 2018-04-15:
Have you solved the issue?

closed 2020-03-19 by xmflsct