niclet/yi-hack-v2 #14: please hack yi v1 upgrade 1080p

pasagame, 2017-01-20:
i have this camera but not working outside china (model 18CN)

link product : https://world.tmall.com/item/543292560507.htm?spm=a220m.1000858.1000725.28.2OL2Kz&id=543292560507&skuId=3433075258232&user_id=1721262917&cat_id=2&is_b=1&rn=65de732e9673836fbc79bfbda964ec61

picture : https://www.mx7.com/view2/zCJiDGq5afEvaDZ0
picture : https://www.mx7.com/view2/zCJjtNikQPngcTif

thank you

tomascrespo, 2017-03-09:
This project is NOT for that camera. You have a similar project for your camera here https://github.com/xmflsct/yi-hack-1080p

Currently there is two Yi Home Camera 1080p. 

- Yi Home Camera v1 upgraded to 1080p 

- Yi Home Camera v2 1080p

xmflsct, 2017-03-11:
@tomascrespo Thanks for the clarification. My project has proper busybox now, but I am still trying to figure out RTSP server stuff. If you have any suggestions, please feel free to share.

Ishforz, 2017-03-31:
Currently there is two Yi Home Camera 1080p.
- Yi Home Camera v1 upgraded to 1080p - https://www.yitechnology.com/yi-1080p-home-camera-2 this?
- Yi Home Camera v2 1080p - https://www.yitechnology.com/yi-1080p-home-camera this?

Hello all!
Someone hacked this camera already https://www.yitechnology.com/yi-1080p-home-camera?

pbuzdin, 2017-06-19:
@Ishforz this is v2 https://www.yitechnology.com/yi-1080p-home-camera-2
and
this is v1 https://www.yitechnology.com/yi-1080p-home-camera
right?