niclet/yi-hack-v2 #15: Intl version of Yi Home 2 ?

macstibs, 2017-02-06:
I successfully flashed back to the older firmware, but the Telnet/FTP/RTSP services aren't available.

Any ideas?

xmflsct, 2017-02-06:
May I ask how did you do that? And which version you were on before downgrading if you can remember? Thanks.

macstibs, 2017-02-06:
I had actually updated through the Yi app to something like 2.1.1_20161206xxxx

Using the files here I got the firmware rolled back to 201604xxxxxx

juliantkp, 2017-04-20:
Hi , how u roll back the firmware? I copied the 2.1.1_20161206xxx n rename to home.bin ... but don seems to be able to upgrade the firmware. Bought the cam recently ...not sure if it is becox latest V2 cam  will not allow us to upgrade. i am outside china. I am using Yi apps china version, 

Currently the cam able to connect with app ...but once lunch it will stuck at 50% (establishing secure channel...50%)  ...then error -20009  :(  bought 4 of this ... now all seating in the box

ykhandler, 2017-07-24:
Check this out:
https://m.youtube.com/watch?v=HssoXD_U7eA

niclet, 2017-07-24:
@ykhandler 
Can you explain a little bit more ?
Do you change the internal serial numbers to be international ?

ykhandler, 2017-07-24:
no, it cannot be changed unless you can hack the server, because there are several id checks. So it still use the Chinese apps.