niclet/yi-hack-v2 #18: Is it possible to use US version in EU region?

nxtpl, 2017-08-22:
Can the chinese proxy trick be adjusted, so it allows US version to work in EU region?
Currently it says "QR code error" while pairing in EU.

niclet, 2017-08-23:
The proxy parameter is not specific to chinese issue, you should be able to use a US proxy the same way.
But I'm not sure that "QR code error" is related to such issue.
When chinese issue occurs, pairing is successful, but watching live video doesn't work.

nxtpl, 2017-08-23:
Thanks for your quick reply.
Both EU and US firmware files are the same, so there must be some IP-related check. Gotta try with proxy and I'll share my findings here.
If there is anything I could help with, I'm offering some testing with my (currently useless) Yi Home Camera 2 EU.

niclet, 2017-08-23:
Yes, there is only one firmware. They are called EU/US and so on, but this is only a download location, not a specific firmware.
Are you on android or iphone ? If on iphone, are you using YiHome from Chinese appstore ?

nxtpl, 2017-08-23:
I'm on iPhone, using the YI Home app. I can also test on Android 6.

Official YI Home app allows me to pair, but only when logged in via US account (which I can't use daily - I already have v1 cams in EU version).

niclet, 2017-08-23:
Ok, so I think it won't work with a US proxy.
I have same issue, I need to be logged in using Chinese account for now.

arf-it, 2017-09-03:
Hi I was doing the same research (same situation: US version to be used in a EU account)..
I think that they are just passing the info taken from the serial number (but this is just my guess).
Issue is that at the moment I don't know how to enable telnet access on an international firmware.

nxtpl, 2017-09-05:
Hi! Thanks for the update. I have trouble simulating a different IP on WLAN connection so I couldn't check what I planned earlier. At the moment I'm using it thru RTSP thanks to your hack.
Yeah, there might be just a simple serial number check VS ip address. I also checked the contents of QR code (maybe there is something more than access point data?), but the strings are encoded.

arf-it, 2017-09-05:
Ok some update: I was able to connect to the camera via serial port. 
Using the terminal it's possible to change the serial number.
Unfortunatelly just changing the string from XXUSYYYY to XXEUYYYY doesn't help so there should be some other parameters used by the camera or it's the server itself that is validating the cam..

niclet, 2017-09-05:
Yes, there is also a key associated to the serial number. And no way to create the correct key for the modified serial :(