niclet/yi-hack-v2 #19: I can't install the hack

tomascrespo, 2017-11-10:
I've followed the instructions. Formated the SD card (FAT32), copy the contents of SD folder into SD card, switch off the camera, insert the SD card, and then power on the camera, but nothings happen. The camera start as always, saving videos to SD card as before, but I don't have telnet access neither FTP access.

I have also reseted the camera with SD inserted, repairing the camera with no succed.

What do I missing?

Thanks. 

My camera is a Yi Home Camera 2, with serial 24CNxxxxxx. Currently I can pair the camera with the chinese app without problem, and the camera works, saving videos to the SD card, but I can't use Live View, it stucks at 50%. I know I also could fix this with your hack, with a china proxy, but currently I can't even get telnet access, so I think the camera is not reading your scripts.

tomascrespo, 2017-11-10:
I think I found the problem. I was editing yi-hack-v2.cfg with notepad. I think notepad change something when save the file (new lines, utf8...) and then the hack doesn't work.
Now I've edited with notepad++ and I have telnet and FTP access 
Yuhuuu!!

However, I can't get rid off 50% stuck problem (even writing a china proxy in YI_HACK_PROXY. I will try with different china proxys

wentianl20, 2018-04-14:
Hi, bro. I have the same model with your device, but I always stuck after reboot the camera with the solid yellow light. Really need your help