niclet/yi-hack-v2 #21: NightVision (IR) does not work when in modified version

mtalexandru, 2017-11-21:
Hello, in modified version the Infrared light is not turning on. If I put the camera in a dark room I cannot see anything on it. 
Running the official firmware, the camera is really OK in dark rooms. 

Anyone else had this problem?

TommyChausson, 2017-11-21:
Hi,

It's a known and unsolved issue : 
https://github.com/niclet/yi-hack-v2/issues/6

mtalexandru, 2017-12-09:
Would you be interested into looking at another similar project which has overcome these issues? The modifications for the Xiaomi MIJIA work really good. They where able to have running rtsp with nightvision and also have cloud at the same time. They also have webserver, samba, ssh, etc.. 

https://github.com/Filipowicz251/mijia-1080P-hacks