niclet/yi-hack-v2 #24: What is your workflow for compiling binaries for this repo?

n8henrie, 2017-12-31:
@niclet has said a few times that he compiles things on the Raspberry Pi:

- https://github.com/niclet/yi-hack-v2/issues/2#issuecomment-241183525
- https://github.com/niclet/yi-hack-v2/issues/12#issuecomment-272144300

Can you give us a basic overview of how this is done?

For example, @shadow-1 has given a [thorough overview of his workflow](https://github.com/shadow-1/yi-hack-v3/issues/23#issuecomment-304882304) for compiling firmware for the v1 cameras -- it would be excellent to have a guide like this for working with the V2 cameras.

For reference, there may be a copy of the ambarella SDK available [here](https://mega.nz/#!kz531D5J!ReQSoOn8LL9bMSw0Z1_5asifpgjMlS17XOg7S8p2Bio), though I can't speak for its authenticity (would be great if someone could verify the sha256; what I'm getting doesn't match the filename.

```shellsession
$ sha256sum ~/Downloads/ambarella-sdk-a0108707d7c5614a702423e63e1e7834009e708a-a0108707
d7c5614a702423e63e1e7834009e708a.tar.gz
0d4d369921147822f4016a4cefa0d2f8c0b85557162283157019db0c3c2eead3  ambarella-sdk-a0108707d7c5614a702423e63e1e7834009e708a-a0108707d7c5614a702423e63e1e7834009e708a.tar.gz
$ sha1sum ~/Downloads/ambarella-sdk-a0108707d7c5614a702423e63e1e7834009e708a-a0108707d7
c5614a702423e63e1e7834009e708a.tar.gz
5152686d45b13e8db9b009457b2a66c3e18dc24a  ambarella-sdk-a0108707d7c5614a702423e63e1e7834009e708a-a0108707d7c5614a702423e63e1e7834009e708a.tar.gz
```

n8henrie, 2018-01-02:
Just tried to see if I could compile a Go binary (since they are statically linked and easy to cross-compile for ARM / Raspberry Pi), but I couldn't find any way to get a "hello world" to work compiling from my Mac using `GOOS=linux GOARCH=arm GOARM=7 go build`, also tried GOARM=6 and 7. Also tried compiling from a Raspberry Pi 3 with those same settings, no luck. Would get "invalid instruction" or a segfault when trying to run the binaries on my yi home 1080 v2.

So then I tried a C++ hello_world compiled on my Raspberry Pi 3: `g++ -static hello_world.cpp -o hello_world`, it also segfaults.

Any suggestions would be much appreciated.

shadow-1, 2018-01-05:
@n8henrie
I suggest biting the bullet and use the cross compiler included with the SDK.

It is a little more difficult to compile applications and libraries compared to compiling natively (usually requires an additional 1 or 2 command line options). However on the positive side, the compiler is specific to the chipset used on the camera and can take advantage of any optimisations specific to the chipset rather than compiling to the lowest common denominator (e.g. armv5).

n8henrie, 2018-01-17:
Thanks for the response.

I will report back to this thread if I'm able to find a walkthrough or any more documentation about the Ambarella SDK -- it looks like it is pretty hard to get ahold of. I sent in a info request at their site and the form response basically says "if you're not considering a large order we probably won't get back to you."

n8henrie, 2018-08-06:
Possibly helpful for future readers:

https://github.com/hello/kasa
https://github.com/hello/kasanojo

n8henrie, 2018-08-28:
Walkthrough for extracting the firmware: https://tech.fpt.com.vn/language/en/extracting-firmware-xiaomi-ants-xiaoyi-smart-camera-part-1/