niclet/yi-hack-v2 #25: Solid Yellow Light on Startup

OmegaNalphA, 2018-03-01:
I've followed the instructions, unplugged the camera, inserted the memory card and re-plugged the camera back in and the camera just has a solid yellow light. It doesn't announce anything or say anything or start blinking to connect, it just sits with the solid yellow light. Has this happened to anyone else? 

I have tried other hacks before this, namely 
https://github.com/shadow-1/yi-hack-v3
https://github.com/fritz-smh/yi-hack
but since mine is the Yi Home Camera 1080P HD V2 I thought this hack would work better.

Thank you!

wentianl20, 2018-04-14:
I have the same issue no matter I disable or enable the China mainland passby. My model is CN only. Really need help !!!

wentianl20, 2018-04-14:
My Serial is 24CN. Probably the model is mot matched with the hack ...orz

meneomic, 2018-12-13:
> I've followed the instructions, unplugged the camera, inserted the memory card and re-plugged the camera back in and the camera just has a solid yellow light. It doesn't announce anything or say anything or start blinking to connect, it just sits with the solid yellow light. Has this happened to anyone else?
> 
> I have tried other hacks before this, namely
> https://github.com/shadow-1/yi-hack-v3
> https://github.com/fritz-smh/yi-hack
> but since mine is the Yi Home Camera 1080P HD V2 I thought this hack would work better.
> Serial is 24US
> 
> Thank you!

I have the same kind of camera (24US)
I tried yi-hack & yi-hack-v2 & yi-hack-v3 ... none of them worked.

Still no news ?