niclet/yi-hack-v2 #31: Current firmware 2.1.0.0E_201809191630

kpj5425, 2019-03-26:
I can't get the firmware to flash at all. I tried the Yi-hack V1, V2, and V3. none of them work. Is this a common problem or has this not been looked at yet? Camera is a Yi 1080p Home Camera.

nelutzu777, 2019-04-18:
Same problem. 
I have Yi 1080p Home Camera, SN: 6FUS...   V1

xfceKris, 2020-03-16:
It's now been nearly a year since your issue posts, and I've got four of these exact same models. Has anyone found a solution?

kpj5425, 2020-03-16:
I don't think anyone is even working on this fix. But since certain updates to HA some of my other devices like switches drop off MQTT everyday and have to restart my router every day to get them back so HA has problems too. I've more or less stopped using my HA program and just turned things on and off via the old fashion way.

tikoPirat3, 2020-07-15:
I need the stock firmware. Could you provide the link please?