niclet/yi-hack-v2 #32: niclet help sdk

delfinogrisu, 2019-04-15:
hi niclet you contact me with the hope that you can give me a hand to take up the project again and solve some problems and create something more besides what you've already done.
it is possible to have complete sdk and related toolchain to be able to continue the development and the creation of the rtsp etc as it is doing crypto with the yi-hack v4 project to which I have already exposed the problem and would also like to integrate the camera to project ...

I invite you to enter the channel and have a chat and tell you about the progress I've already made with this room ;-)
https://discord.gg/upPsFWZ

ghiglie, 2019-08-25:
I've found an Ambarella SDK here: https://gitlab.com/hugowhiteflame/ambarella-sdk/ but I have no idea on how to move on.

delfinogrisu, 2019-08-25:
thank you very much ... are you Italian too ???  I passed it all to crypto yi hack v4 and it told me that it doesn't find the cross compiler in the package I hope it can create something to be able to insert also our room in its project ... thanks again Marco !!!

ghiglie, 2019-08-25:
Yes, I am.
Also found and tried this: https://techinsight.com.vn/language/en/extracting-firmware-xiaomi-ants-xiaoyi-smart-camera-part-1/
I can extract the UBI image from the firmware - again, I don't know how to go on.
Good news if Crypt0 can add this cam to the v4! it's totally a different platform anyway.